cqparts/cqparts #139: cqparts parameter access

tolot27, 2018-12-28:
I'd like to use a class parameter as the template param of the render_props function and defined it as following:

```python
class Case(cqparts.Part):
    # Parameters
    color = String("green", doc="color-coded position")
    print(color)
    _render = render_props(template=color)
```

Unfortunately, it is not accessed like a str as it is expected by the template param and can be seen with the print statement (it prints: `<cqparts.params.types.String object at 0x000002602F2ACBE0>`). If I call `render_props(template=color.default)` it works but uses the default value and not the passed value. How to access the value passed as the class parameter?

In `cqparts/tests/t_cqparts/test_params.py` only default and casts are tested.

BTW: Using native types like `str` do not work as parameters.

CadQuery 1.2.0

tolot27, 2018-12-28:
It looks like it is only possible to get the value via self. Hence I changed it like this:

```python
class Case(cqparts.Part):
    # Parameters
    color = String("green", doc="color-coded position")
    def initialize_parameters(self):
        self._render.color = COLOR[self.color]
```

tolot27, 2018-12-29:
Unfortunately, my workaround does not work with multiple shapes based on the same class. It only works if I use `RenderParam` as described in [cqparts.display.material.RenderProp](https://cqparts.github.io/cqparts/doc/api/cqparts.display.html?highlight=renderprops#cqparts.display.material.RenderProps). But that decreases the readability.

fragmuffin, 2018-12-30:
@tolot27 : yeah, that's not supposed to happen.

```python
from cqparts_misc.basic.primatives import Box
from cqparts.params import *
from cqparts.display.material import COLOR

class Case(Box):  # inherits from Box just to give it shape
    color = String('red')
    def initialize_parameters(self):
        self._render.color = COLOR[self.color]

# Create Objects
c1 = Case(color='green')
c2 = Case(color='blue')
```

So far so good, however:

```python
>>> id(c1._render) == id(c2._render)
True
```

So the instances `c1` and `c2` contain the same `cqparts.display.material.RenderProps` instances for their `_render` property.
That's *not* intended, these properties are meant to be unique per `Part`, similar to a length, or a radius.

I'm looking into it.
Sorry for the delay, Christmas, and now I have a massive headache (not a hangover :wink: ).

fragmuffin, 2018-12-30:
OK! here's the problem: https://github.com/cqparts/cqparts/blob/7135b4dcaa5b80aa785b1ed301463377ad711e5b/src/cqparts/params/parametric_object.py#L71
This assigns *the* default `RenderProps` instance (intended for all objects) to the instance currently being made.
This code was written with numbers in mind, like `length` or `radius` (as mentioned above)... however, `RenderProps` being an object has internal properties like `color` that, when altered, effect the properties of *all* default rendered parts.
This bug could also easily extend to other more complex parameters intended for a `Component`.

tolot27, 2018-12-30:
It works now as expected. Many thanks!