cqparts/cqparts #148: Remove FreeCAD dependency

bernhard-42, 2019-01-27:
I have added cqparts support for my jupyter extension [https://github.com/bernhard-42/cadquery-jupyter-extension](https://github.com/bernhard-42/cadquery-jupyter-extension). Since the extension runs only with the pythonOCC version of cadquery, I had to patch `geometry.py`

```
 class CoordSystem(cadquery.Plane):
     """
     Defines the location, and rotation of an orthogonal 3 dimensional coordinate
@@ -108,14 +103,14 @@ class CoordSystem(cadquery.Plane):
             i & = cos(\beta) cos(\gamma)
         """
         # Create reference points at origin
-        offset = FreeCAD.Vector(0, 0, 0)
-        x_vertex = FreeCAD.Vector(1, 0, 0)  # vertex along +X-axis
-        z_vertex = FreeCAD.Vector(0, 0, 1)  # vertex along +Z-axis
+        offset = cadquery.Vector((0, 0, 0))
+        x_vertex = cadquery.Vector((1, 0, 0))  # vertex along +X-axis
+        z_vertex = cadquery.Vector((0, 0, 1))  # vertex along +Z-axis

         # Transform reference points
-        offset = matrix.multiply(offset)
-        x_vertex = matrix.multiply(x_vertex)
-        z_vertex = matrix.multiply(z_vertex)
+        offset = offset.transform(matrix)
+        x_vertex = x_vertex.transform(matrix)
+        z_vertex = z_vertex.transform(matrix)

         # Get axis vectors (relative to offset vertex)
         x_axis = x_vertex - offset
```

I cannot test against the FreeCAD version since on macos there are issues around python versions between FreeCAD and system python (one of the reasons for me to leave FreeCAD)

Thought it could maybe help ...

bernhard-42, 2019-01-28:
Looking forward to the next release and will switch over to feature/freecad-occ for the time being

jhselvik, 2020-08-16:
I too am looking forward to this release, thank you for your work. 

This may be a dumb question, but I do not see a `setup.py`, how do you install the `feature/freecad-occ` branch to use in a project? We are using cadquery 2 and would like to avoid including FreeCAD in our environment.

bernhard-42, 2020-08-16:
@jhselvik You might look into https://github.com/cqparts/cqparts/blob/master/deployment/deploy.sh It seems to create the `setup.py` file, see also https://github.com/cqparts/cqparts/issues/149

jhselvik, 2020-08-16:
Thank you @bernhard-42, I will give that a try and let you know how it works!