cqparts/cqparts #149: Adapt build script for Mac

bernhard-42, 2019-01-27:
Macos' foundation is some BSD and hence `find` does not have `-printf` flag

I had to install `findutils`
```
brew install findutils
```
and set
```
export FIND=gfind
```
and patch `deployment/deploy.sh`

```
diff --git a/deployment/deploy.sh b/deployment/deploy.sh
index e160f3c..7e01d5b 100755
--- a/deployment/deploy.sh
+++ b/deployment/deploy.sh
@@ -5,6 +5,7 @@ CONTAINER_NAME=deploytest

 CQPARTS_ROOT=$(git rev-parse --show-toplevel)

+FIND=${FIND:-find}

 function show_help() {
     [ "$@" ] && echo "$@"
@@ -48,12 +49,12 @@ Arguments:

 Libraries Available for Deployment: {lib}
 EOF
-find ../src -maxdepth 1 -type d -name "cqparts*" -printf "    %P\n" | sort | grep -vP '^(.*\.egg-info|\s*)$'
+$FIND ../src -maxdepth 1 -type d -name "cqparts*" -printf "    %P\n" | sort | grep -vP '^(.*\.egg-info|\s*)$'
 cat << EOF

 Environments: {env}
 EOF
-find env -maxdepth 1 -type d -printf "    %P\n" | sort | grep -vP '^\s*$'
+$FIND env -maxdepth 1 -type d -printf "    %P\n" | sort | grep -vP '^\s*$'

 }
```

Not sure you're interested in supporting Mac for builds (and I haven't tested the whole docker stuff).
However, maybe some macos user might find it helpful

fragmuffin, 2019-01-28:
Hi @bernhard-42 

Thanks for the heads up, I wasn't aware that mac's `find` command didn't support `printf`.

However the `deploy.sh` script is _only_ for an author/maintainer to release a version of `cqparts` to PyPI (`pip`). (all written in the [`deployment/README.md`](https://github.com/cqparts/cqparts/blob/master/deployment/README.md))

And since it's just me at this time, and I don't use a mac, there's no need for that script to be multi-platform.

> Not sure you're interested in supporting Mac for builds

No, not for _building_ packages, however I _am_ interested in testing it as part of CI, so you've reminded me to add `ios` to TravisCI.

Thanks @bernhard-42 !

closed 2019-01-28 by fragmuffin

fragmuffin, 2019-01-28:
@bernhard-42 I was just thinking... why are you running this script?
do you need to build `cqparts` as part of your development process for [`bernhard-42/cadquery-jupyter-extension`](https://github.com/bernhard-42/cadquery-jupyter-extension) ?

If _you_ need to run `deploy.sh` on your system, then please create a PR with those changes working on your mac, and I'll merge when I can see it still working on mine.

bernhard-42, 2019-01-28:
Needed setup.py for my  FreeCAD removals (wasn't aware of the branch you mentioned in the other issue). As soon as the new version without FreeCAD is public, I won't need it any more - I guess ;-)
Nevertheless, will come up with a PR, maybe slightly smarter without brew installation.

bernhard-42, 2019-01-28:
Would it be an option to just create and then keep the latest `setup.py` in the repo?
Then one could change some python code in his own fork and `pip install` it without bothering about the `deploy.sh` script