cqparts/cqparts #150: How to get rid of "solving for Assembly without world coordinates" warning

bernhard-42, 2019-01-27:
When I render `cqparts` models I get the warning `solving for Assembly without world coordinates ...`
Not sure what I am doing wrong. In my code I create a `cqparts` model, e.g. your toy car in var `car`. The `display` routine then calls `car.solve()` and then recursively converts the `cqparts.Assembly` into an `cq_jupyter.Assembly` which then gets shown.

Am I'm missing something?

fragmuffin, 2019-01-28:
Hi @bernhard-42 

All you're doing is building something with the assumption that its origin is at the world's origin.
The solution is to remove that assumption by telling the assembly where it is.

```python
>>> from cqparts_bearings.ball import BallBearing

# Replicating what you're seeing
>>> obj1 = BallBearing()
>>> obj1.solve()
solving for Assembly without world coordinates set: ...

# Solve by telling the object where it is in the world
>>> from cqparts.utils import CoordSystem
>>> obj2 = BallBearing()
>>> obj2.world_coords = CoordSystem()  # defaults to 0,0,0
>>> obj2.solve()  # no warning shown
```

for reference: [`CoordSystem` documentation](https://cqparts.github.io/cqparts/doc/api/cqparts.utils.html#cqparts.utils.geometry.CoordSystem)

**Why the warning?**
It's there as a crude safeguard against pre-maturely solving sub-assemblies.
For example, for an assembly:

* `arm`
    * `hand`

solving `arm` will give `hand` a `world_coords`, and consequently any `fingers` will get their coordinates offset from the hand's origin (presumably the wrist)

However, if `arm` was part of a larger object:

* `person`
    * `arm`
        * `hand`

Now we have a `hand` who's world coordinates did not take into account the offset presented by the `arm` (because we assumed it was at the origin).

**"That _is_ crude"**
Yeah, I know... recursively solving for `world_coords` _should_ be properly safeguarded and raise an exception when it's wrong, making it ok to assume something is at `0,0,0` if not explicitly set.
There are major changes I'm looking at making to that whole structure (#107 as an example) which fundamentally changes how all this works.

closed 2019-01-28 by fragmuffin

bernhard-42, 2019-01-28:
cool, added it. Thanks