cqparts/cqparts #152: cqparts.Assembly does not have a bounding_box function

zignig, 2019-02-15:
cqparts.Part can calculate a bounding_box by creating a solid and returning the BB of it. Unfortunately the cqparts.Assembly does not have an equivalent  function. 

A bounding box function would be useful for my cqparts_webapi, where I want to create preview images of a collection of Assemblies.  I would collect the bounding boxes of the Assemblies and them place them with a gap based on the bounding box. 

A recursive addition of the individual boxes would probably do it , but i am not sure where to start .

fragmuffin, 2019-02-15:
Good point! I can do that!
But when to get the time 😢

zignig, 2019-02-15:
The primary reason for this issue is that I saw all  77 of the cqparts that I have made in a single gltf.

I would take about 1/2 an hour to build, but all of it W00t!.

fragmuffin, 2019-02-17:
closed with #153

fragmuffin, 2019-02-17:
Thanks for the feedback @zignig.
A bounding box for `Assembly` instances shall be very useful indeed.

closed 2019-02-17 by fragmuffin