cqparts/cqparts #154: Make internal GLTF Convertor for CAD models that based on cqparts

mrnexeon, 2019-03-21:
Hello, today is a problem that there’s no GLTF converters for CAD formats like *.stp. 

The cqparts can be a good candidate for that purposes, but cqparts has a heavy dependencies like FreeCAD and CadQuery, can we do the convertor without them?

fragmuffin, 2019-03-23:
Welcome @mrnexeon !

It's not possible to use `cqparts` without `cadquery`. That's because the fundamental purpose of `cqparts` is to be a wrapper around `cadquery`... and a wrapper without the insides is just rubbish :wink: 

But you're absolutely right about FreeCAD, there's been an ongoing initiative to use `python-occ` instead of `FreeCAD` to remove it as a dependency!

cadquery is being moved to: https://github.com/cadquery/cadquery where it's readme says:

> This is an experimental CadQuery fork that uses PythonOCC instead of FreeCAD. It also has preliminary Jupyter notebook integration.
> 
> You can try it out on linux-64, osx-64 and win-64 using conda:
> ```
> conda install -c pythonocc -c oce -c conda-forge -c dlr-sc -c CadQuery cadquery-occ
> ```

But you'll also need an equally experimental branch of `cqparts`

```bash
git clone git@github.com:cqparts/cqparts.git
cd cqparts
git checkout feature/freecad-occ
PYTHONPATH=${PWD}/src/cqparts:$PYTHONPATH
```

To check your python environment, you should (at the time of writing this) see the versions:

```python
>>> import cadquery, cqparts
>>> cadquery.__version__
'1.0.0'
>>> cqparts.__version__
'0.2.2.dev1'
```

Then you should be able to import `STEP` and export to `glTF v2.0` with:

```
>>> import cqparts
>>> cqparts.Assembly.importer('step')('myfile.stp').exporter('gltf')('myfile.gltf')
```

mrnexeon, 2019-03-23:
@fragmuffin Wow! That’s a cool, thanks!

closed 2019-03-23 by mrnexeon

mrnexeon, 2019-03-27:
@fragmuffin, what if there’s only report message in terminal and no one gltf as output?
[image]

fragmuffin, 2019-03-29:
I haven't seen that before.
Would you be able to share your `gearbox.stp` file?