cqparts/cqparts #159: PROJECT FUTURE

Andrei-Pozolotin, 2020-03-19:
* there was an abrupt drop in project activity
* I am curious if this project is dead / abandoned?

fragmuffin, 2020-03-20:
Hi @Andrei-Pozolotin , this project is not dead... but it has been very difficult for me to spend much time on it.
There will be a resurgence on this project one day, I just don't know when that will be.

closed 2020-03-20 by fragmuffin

Andrei-Pozolotin, 2020-03-21:
@fragmuffin Peter:

1. thank you for the update

2. cqparts currently depends on FreeCAD all over the place
https://github.com/cqparts/cqparts/search?q=FreeCAD
which is a pain when working with CQ2:
https://github.com/CadQuery/cadquery
which is using OCCT instead of FreeCAD

3. do you plan to migrate to CQ2?

jmwright, 2020-03-21:
@Andrei-Pozolotin I've got a fork where the tests pass with CQ 2.0, but it's just a rough pass at getting things to work. I'm not using it for anything currently, so I don't know how well it works.

https://github.com/jmwright/cqparts

I can't speak to the intention of the core developers with respect to CQ 2.0.

Andrei-Pozolotin, 2020-03-21:
@jmwright Jeremy:
* wow, thank you, I will study your fork
* do you publish artifacts to conda? or...?

Andrei-Pozolotin, 2020-03-21:
FYI:
one more fork trying to get rid of FreeCAD
https://github.com/everlanes/cqparts/commits/master

jmwright, 2020-03-21:
If you start at this comment and work backwards you can get some more context. https://github.com/cqparts/cqparts/issues/129#issuecomment-588504842

There's nothing published. And to be honest, I'm not very familiar with publishing packages for conda. Adam has done pretty much all of that for the CQ repo.

A couple things to note.

1. My update is based on the OCP bindings that Adam is working on. A link to a pre-compiled binary for Linux is in a comment on the issue above. The OCP and PythonOCC bindings should be very similar for OCCT 7.4, but I haven't done a comparison.

2. I haven't modified the original cqparts CI configuration to work properly. You'll have to build the CQ environment based on the instructions in that repo's readme, followed by a manual running of the tests. Once I have the CQ environment built I just add the cqparts directory to PYTHONPATH and then it's available for use from the command line. It doesn't have any compiled dependencies other than the OCCT/PythonOCC/CQ layer that I'm aware of.

I also haven't fixed cqparts to run in CQ-editor as an environment either, so trying to run a cqparts script from within CQ-editor won't work currently (unfortunately).

I don't have any plans for the fork, but if you're interested in working on it we should probably start a new issue here for that discussion so that it's separate and people can just ignore that issue if they're not interested.

Andrei-Pozolotin, 2020-03-21:
@jmwright Jeremy:

1. thank you for the context, I will study that

2. on the issue of OCCT 7.4.0, please take a look
https://github.com/CadQuery/cadquery/pull/300
can you please voice your opinion:
https://github.com/CadQuery/cadquery/issues/226#issuecomment-601867495
* should there be a branch?
* should just abandon 0.18 in favor of 7.4?
* etc, etc.

thank you

zignig, 2020-03-22:
@Andrei-Pozolotin  if you want to have a collection of cq_parts to browse and GROK, 

https://github.com/zignig/cqparts_bucket

processed as a website and rendered and built as gltf

https://cqparts.github.io/examples/

Andrei-Pozolotin, 2020-03-22:
@zignig Simon:
very cool idea, thank you

Andrei-Pozolotin, 2020-03-23:
@jmwright Jeremy:

FYI:

based on your work:
https://github.com/random-builder/design_cadquery_parts/commit/acf817c49a4492da51af3ea9c8a22a0ddcadb4c6

I published conda cqparts bundle 
https://anaconda.org/random-builder/cq-parts-bundle

with install snippet for cq + cqparts:
```
conda install -y \
    -c conda-forge \
    -c cadquery cadquery=2 \
    -c random-builder cq-parts-bundle \
```

so far seems to work fine with current (occt-0.18 based) CQ2

jmwright, 2020-03-23:
@Andrei-Pozolotin Thanks for the work you've put into it.