cqparts/cqparts #162: Is it possible to somehow name elements within a STEP file or otherwise assign exportable attributes using CadQuery?

chendy, 2020-04-14:
So I don't know anything really about STEP files, CadQuery or cqparts.

I want to define an assembly, perhaps like [here](https://cqparts.github.io/cqparts/doc/tutorials/assembly.html). I want to tag the chassis as 'chassis' and wheel as 'wheel'. So that hopefully I can setup blender to render the chassis as a metal material, and the wheel as rubber, using a script.

Or otherwise, I want to have a fully command line workflow, where I design assemblies of parts of different materials, and somehow render to a reasonable quality. The link above, where the car had different colour elements is below expectations. 

I was thinking blender would be a natural compliment to CadQuery for this task, but I'm a complete noob. 

I posted here after [suggestion from Jeremy Wright](https://groups.google.com/forum/#!msg/cadquery/zwgR0pTv1_g/Rye_fiB8AAAJ)

@zignig

jmwright, 2020-04-14:
Just to repost some of what I said on the Google Group and add a little more detail.

1. There is not a cqparts release that's build on CQ 2.0 yet, but some work has been done in that direction.
2. jupyter-cadquery has lightweight assemblies as well. However, I think cqparts assembly functionality has some more depth.
3. cqparts has good gltf support I believe, which may make importing to Blender easier. https://github.com/KhronosGroup/glTF-Blender-IO
4. Tagging has been implemented in CQ 2.0 and is documented [here](https://cadquery.readthedocs.io/en/latest/examples.html#tagging-objects). However, that information is not translated by the exporter. That's something that I would like to see be implemented though eventually.
5. CQ does not support colors, materials, etc as part of an object's metadata currently. I think that cqparts at least does colors though.

jmwright, 2020-05-02:
@chendy You might have a look [here](https://github.com/CadQuery/cadquery/issues/276#issuecomment-585967939) for a reference on how light-weight assemblies are done in jupyter-cadquery.