cqparts/cqparts #163: Temporary installation instructions

freeflydeveloper, 2020-05-06:
I understand from reading other issue threads that this project is transitioning to cadquery 2.

I'm curious if there is any way to use it in the interim. I've not been successful with installing a legacy version of cadquery and the experimental [cq-parts-bundle](https://anaconda.org/random-builder) failed to install on mac (cadquery-pyocct was unavailable).

My use case is batching conversion of step files to gltf.

jmwright, 2020-05-06:
If you're ok with using CadQuery 1.x, cqparts should still work in the FreeCAD workbench. https://github.com/jmwright/cadquery-freecad-module/tree/master/ThirdParty

You should be able to install it through the FreeCAD add-ons manager.

alekssadowski95, 2021-08-20:
Does cqparts work from the FreeCAD Cadquery addon? 
I am using it like this:
```
import cqparts
cqparts.Assembly.importer('step')('myfile.stp').exporter('gltf')('myfile.gltf')
```
but I get:
`AttributeError: 'Shell' object has no attribute 'tesselate'`

jmwright, 2021-08-23:
@alekssadowski95 cqparts has a custom gltf exporter because CadQuery didn't support gltf at the time cqparts was created (the CQ gltf export wasn't added until after the CadQuery 2.1 release). I'm not sure how the custom exporter works though, so I'm not sure how well it will play with the workbench.

alekssadowski95, 2021-08-23:
I dont want to use the custom exporter. I want to export to gltf.

jmwright, 2021-08-23:
The custom exporter that is included with cqparts does export to gltf, it's just that the cqparts team just implemented it themselves. If you want to use the GLTF export that is built-in to CadQuery now, you will need to move to CQ-editor and a newer version of CadQuery. The FreeCAD workbench does not support newer versions of CadQuery. I also don't think that cqparts is supported any more, although one of their team members please correct me if I'm wrong.

CQ-editor: https://github.com/CadQuery/CQ-editor

alekssadowski95, 2021-08-23:
Didnt know that cadquery 2 exports to gltf. as its not mentioned in the online documentation. Can cadquery 2 export to X3D?

jmwright, 2021-08-23:
Sorry, I forgot that there is a small issue holding up GLTF support. I thought it had already been merged.

X3D is not supported. https://cadquery.readthedocs.io/en/latest/importexport.html