cqparts/cqparts #164: Conversion from STEP to GLTF - "tessellate" error

Horytnik, 2021-02-26:
Hello, 
I am trying to convert step file to gltf file. According to following issue I was trying to use cqparts and importers. 
https://github.com/cqparts/cqparts/issues/72#issuecomment-424605083

I pulled docker image with FreeCAD installed from there
https://hub.docker.com/r/amrit3701/freecad-cli

Then I installed cqparts and run following code
```
from cqparts import Assembly

Assembly \
    .importer('step')('Test.step') \
    .exporter('gltf')('output/test.gltf')
```
In output folder i got several .bin files and following error

`WARNING: CadQuery 1.x is no longer supported. Please upgrade to CadQuery 2.x https://github.com/CadQuery/cadquery
Traceback (most recent call last):
  File "convert.py", line 16, in <module>
    Assembly \
  File "/usr/local/lib/python3.8/dist-packages/cqparts/codec/gltf.py", line 430, in __call__
    add(
  File "/usr/local/lib/python3.8/dist-packages/cqparts/codec/gltf.py", line 407, in add
    add(
  File "/usr/local/lib/python3.8/dist-packages/cqparts/codec/gltf.py", line 417, in add
    self.add_part(
  File "/usr/local/lib/python3.8/dist-packages/cqparts/codec/gltf.py", line 584, in add_part
    buff = self.part_buffer(part)
  File "/usr/local/lib/python3.8/dist-packages/cqparts/codec/gltf.py", line 528, in part_buffer
    (vertices, indices) = cls.part_mesh(part)
  File "/usr/local/lib/python3.8/dist-packages/cqparts/codec/gltf.py", line 492, in part_mesh
    tess = shape.tessellate(cls.tolerance)
AttributeError: 'Shell' object has no attribute 'tessellate'`

Next, according the following issue I installed anaconda on docker but I can't import this cqparts. 
https://github.com/cqparts/cqparts/issues/154

My PYTHONPATH looks like below. I checked out the feature/freecad-occ
`(py385) root@1baafabb15e6:/convert/output# echo $PYTHONPATH
/convert/cqparts/src/cqparts/:/usr/local/lib:`

When I installed cqparts by pip again I got the same error as previous
`WARNING: CadQuery 1.x is no longer supported. Please upgrade to CadQuery 2.x https://github.com/CadQuery/cadquery
Traceback (most recent call last):
  File "conversion.py", line 7, in <module>
    Assembly \
  File "/root/anaconda3/envs/py38/lib/python3.8/site-packages/cqparts/codec/gltf.py", line 430, in __call__
    add(
  File "/root/anaconda3/envs/py38/lib/python3.8/site-packages/cqparts/codec/gltf.py", line 407, in add
    add(
  File "/root/anaconda3/envs/py38/lib/python3.8/site-packages/cqparts/codec/gltf.py", line 417, in add
    self.add_part(
  File "/root/anaconda3/envs/py38/lib/python3.8/site-packages/cqparts/codec/gltf.py", line 584, in add_part
    buff = self.part_buffer(part)
  File "/root/anaconda3/envs/py38/lib/python3.8/site-packages/cqparts/codec/gltf.py", line 528, in part_buffer
    (vertices, indices) = cls.part_mesh(part)
  File "/root/anaconda3/envs/py38/lib/python3.8/site-packages/cqparts/codec/gltf.py", line 492, in part_mesh
    tess = shape.tessellate(cls.tolerance)
AttributeError: 'Shell' object has no attribute 'tessellate'`

For now I am completely stuck and I don't know how to fix it. Could anyone help me?
Best Regards
Michal

zignig, 2021-02-28:
unfortunately cqparts h a s gotten out of sync with Cadquery.  If you are will to bring it we can fix it together @fragmuffin .

Horytnik, 2021-03-01:
It would be really nice because for now I haven't found any other way to load step files or any other typical file which contains the assembly structure into Three JS. The conversion from step to gltf would solve this problem. Thank you :)

Horytnik, 2021-03-11:
Hello @zignig one more thing. Maybe there is a way to somehow download older versions which were synchronized? If not then how can I help with this sync?

Spoutnik97, 2023-08-02:
Hi!
any news about this thread ? I'm interested 🙂

Farouk-Belhassine, 2023-08-08:
i'm using the same method and getting another error in the export, i get "module 'cadquery' has no attribute 'freecad_impl'"