cqparts/cqparts #166: What's the future of this library - trouble installing ?

MarcWeber, 2022-10-09:
Assemblies: https://cadquery.readthedocs.io/en/latest/assy.html
take over part of whats included here.
https://readthedocs.org/projects/cq-warehouse/ seems to be an alternative library for components.
And on discord I was told that cqparts might be broken and no longer maintained ? I had trouble installing it.

So if you're just looking for coincidence like code checkout the CQ assemblies (?)

g-simmons, 2022-12-15:
@MarcWeber Seems like cqparts was broken when cadquery went to major version 2, switching from FreeCAD to OCC backend. It looks like fixing cqparts to be compatible with cadquery 2 was almost completed at some point? #129

g-simmons, 2022-12-15:
Looks like updated version of cqparts is on the branch referenced in this PR https://github.com/cqparts/cqparts/pull/155